kuoruan/openwrt-v2ray #101: wrt1900ac v2ray plug-in

taricano, 2020-10-01:
hi, I am new to openwrt and I do not speak Chinese. I am looking for a plugin to install on my openwrt interface that would setup a list of my servers automatically such as the fancyss for AsusMerlin or the shadowsocks plus for the raspberry Pi. 

I have already installed the kuoruan v2ray into my router but I have no clue how to set it up and the guides I read in Chinese do not translate to english properly. 

any help would be greatly appreciated.

kuoruan, 2020-10-03:
Have you tried my luci-app-v2ray?

taricano, 2020-10-03:
Yes, but I don’t know how to configure my servers in it to work properly. I
am looking for something where I can copy and paste my servers or a
subscription link for everything to be added automatically like the
shadowrocket or V2rayN app/software or some other plug-ins available in
some firmware.

kuoruan, 2020-10-03:
You can use the `Outbound Import`, and paste the urls which copy from your APP.

taricano, 2020-10-03:
What exactly should I do after importing the servers in the outbound tab?

What is the file v2ray.gz.tar for?