kuoruan/openwrt-v2ray #103: How to force router itself to go through transparent proxy?

Tualua, 2020-10-08:
I'd like to route all router traffic through my v2ray server.
Now, if I check with `curl 2ip.ru` it gives me v2ray server address but for example zerotier service keeps going through default pppoe connection.

Is there any way to force all router traffic to v2ray?

kuoruan, 2020-10-09:
for some reason, we can't get router itself to go through transparent proxy at this time.
we are looking for some way to resolve this.

taricano, 2020-10-09:
Does the router that you run the test on 2ip.ru connected directly to the ISP cable through pppoe?

Tualua, 2020-10-09:
Yes

taricano, 2020-10-09:
instead, connect a generic router to your ISP and set the DNS 1 and 2 on the WAN to use YANDEX public DNS, then plug in the router you are currently using to it but do not set as a secondary device. For example, if the first router IP is 192.168.1.1 set the other one as 192.168.2.1. on LAN, then change the WAN interface to use DHPC (try to set the DNS 1 & 2 to google for example.) After this is complete, run a test on dnsleaktest.com to see if there is a leak.

vagabund89, 2021-04-23:
is there a way to send all traffic through v2ray?

arinc9, 2021-07-25:
@Tualua @vagabund89 are you still in the need of doing transparent proxy? I'm making a guide for that, let me know if you'd like to see it.

vagabund89, 2021-07-25:
Yes I need it

arinc9, 2021-07-25:
Take a look at TPROXY. You need to install iptables-mod-tproxy package and configure dokodemo-door as inbound on v2ray. The rest is pretty straightforward.

https://www.notion.so/arinc9/Transparent-Proxy-Using-v2ray-c6bfee7be82f4084a73a132e362c6a10