kuoruan/openwrt-v2ray #109: 编译报错,4.28-2版本后,都报错

TerryWang159, 2020-11-06:
在Openwrt官方源代码 编译 
V2ray-core 最新版 编译报错 如下 . 仔细测试了一下 4.28-2版本后都编译不通过 , 回滚到4.28-2就可以编译成功 . 
报错信息  :
Makefile:350: recipe for target '/home/terry/openwrt/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-4.31.3/.built' failed
make[2]: *** [/home/terry/openwrt/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-4.31.3/.built] Error 2
make[2]: Leaving directory '/home/terry/openwrt/package/v2ray-core'
time: package/v2ray-core/full/compile#72.47#24.94#40.67
package/Makefile:111: recipe for target 'package/v2ray-core/compile' failed
make[1]: *** [package/v2ray-core/compile] Error 2
make[1]: Leaving directory '/home/terry/openwrt'
/home/terry/openwrt/include/toplevel.mk:225: recipe for target 'package/v2ray-core/compile' failed
make: *** [package/v2ray-core/compile] Error 2

kuoruan, 2020-11-06:
Duplicate of #108

closed 2020-11-06 by kuoruan

wukaisky, 2021-05-03:
openwrt 19.07需要升级golang，方法如下：

diff --git a/lang/golang/golang-version.mk b/lang/golang/golang-version.mk
index f3bd0957a..7f4e1a1ff 100644
--- a/lang/golang/golang-version.mk
+++ b/lang/golang/golang-version.mk
@@ -10,5 +10,5 @@ ifeq ($(origin GO_INCLUDE_DIR),undefined)
@@ -10,5 +10,5 @@ ifeq ($(origin GO_INCLUDE_DIR),undefined)
 endif
 
 
-GO_VERSION_MAJOR_MINOR:=1.13
-GO_VERSION_PATCH:=15
+GO_VERSION_MAJOR_MINOR:=1.16
+GO_VERSION_PATCH:=3
diff --git a/lang/golang/golang/Makefile b/lang/golang/golang/Makefile
index c4857fc29..6efe8131b 100644
--- a/lang/golang/golang/Makefile
+++ b/lang/golang/golang/Makefile
@@ -18,7 +18,7 @@ GO_SOURCE_URLS:=https://dl.google.com/go/ \
 
 PKG_SOURCE:=go$(PKG_VERSION).src.tar.gz
 PKG_SOURCE_URL:=$(GO_SOURCE_URLS)
-PKG_HASH:=5fb43171046cf8784325e67913d55f88a683435071eef8e9da1aa8a1588fcf5d
+PKG_HASH:=b298d29de9236ca47a023e382313bcc2d2eed31dfa706b60a04103ce83a71a25
 
 PKG_MAINTAINER:=Jeffery To <jeffery.to@gmail.com>
 PKG_LICENSE:=BSD-3-Clause