kuoruan/openwrt-v2ray #112: R2S 应该用哪个核心？

infisz, 2020-11-18:
正在执行软件包管理器
Unknown package 'v2ray-core'.
错误
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package v2ray-core.

版本和系统不对应，不能安装，A53 A7,都试过了。小白求助。

kuoruan, 2020-11-20:
贴出信息：
```
cat /etc/*release
opkg print-architecture
```

lyx32, 2020-12-08:
root@FriendlyWrt:~# cat /etc/*release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_TARGET='rockchip-rk3328/generic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_TAINTS='no-all'
DISTRIB_REVISION='R20.7.1'
DISTRIB_DESCRIPTION='OpenWrt '
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r2993+618-b9a408c2b4"
OPENWRT_BOARD="rockchip-rk3328/generic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r2993+618-b9a408c2b4"
root@FriendlyWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch aarch64_cortex-a53 10

kuoruan, 2020-12-08:
@lyx32 `aarch64_cortex-a53`