kuoruan/openwrt-v2ray #114: 请问如何降低 System Log 中 v2ray 的 log level？

h2fishcake, 2020-11-25:
已在 v2ray 的配置中设置了 `"log":{"loglevel":"warning"}`  
但System Log中还是会出现如下图大量的 access log (V2Ray 4.32.1)  
[image]

Mattraks, 2020-12-03:
[LogObject](https://www.v2fly.org/config/overview.html#logobject)

h2fishcake, 2020-12-03:
> [LogObject](https://www.v2fly.org/config/overview.html#logobject)

没看问题内容吗？
已经设置了loglevel，但似乎影响不了system log里出现的的v2ray

Mattraks, 2020-12-03:
> > [LogObject](https://www.v2fly.org/config/overview.html#logobject)
> 
> 没看问题内容吗？
> 已经设置了loglevel，但似乎影响不了system log里出现的的v2ray

你是没看到第一行的吗?
 "access": "none"

h2fishcake, 2020-12-03:
可以了，感谢👍👍

closed 2020-12-03 by h2fishcake