kuoruan/openwrt-v2ray #119: Xiaomi Nano R1CL mipsel_24kc OpenWRT 19.07.5

alcatraz-zz, 2021-01-25:
Please attach the output of following commands.

```
cat /etc/*release
opkg print-architecture
```

cat /etc/*release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.5'
DISTRIB_REVISION='r11257-5090152ae3'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.5 r11257-5090152ae3'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.5"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.5"
VERSION_ID="19.07.5"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r11257-5090152ae3"
OPENWRT_BOARD="ramips/mt76x8"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.5 r11257-5090152ae3"
root@OpenWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10

kuoruan, 2021-01-25:
是跑不起来吗？

palorotolo, 2022-02-24:
同样，安装提示：
 * pkg_hash_check_unresolved: cannot find dependency ca-certificates for v2ray-core-mini
 * pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core-mini found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package v2ray-core-mini.