kuoruan/openwrt-v2ray #121: MIPS 74Kc

mkmark, 2021-02-21:
Please attach the output of following commands.

```
root@TP-LINK:/etc# cat /etc/*release
DISTRIB_ID="OpenWrt"
DISTRIB_RELEASE="Barrier Breaker"
DISTRIB_REVISION="r60685"
DISTRIB_CODENAME="barrier_breaker"
DISTRIB_TARGET="ar71xx/generic"
DISTRIB_DESCRIPTION="OpenWrt Barrier Breaker 14.07"
DISTRIB_TAINTS="no-all no-ipv6 busybox"

root@TP-LINK:/etc# opkg print-architecture
arch all 1
arch noarch 1
arch ar71xx 10
```

My router TL-R473G is based on OpenWRT but is not openly supported. I managed to gain the root access.

The CPU model information is unclear as it's not provided by OEM. A quick search on google returns [results](https://www.ruten.com.tw/item/show?22002183189152) showing that it's using a 775Mhz processor, which combined with MIPS gives search results showing that it is most likely using QCA9563, or QCA956X, both are MIPS 74Kc.

All the compiled MIPS version provided won't run.

I tried to compile the ipk with the latest sdk, and succeeded, but v2ray won't run. ldd shows

```
root@TP-LINK:~# ldd v2ray
ldd: can't open cache '/etc/ld.so.cache'
checking sub-depends for '/lib/libgcc_s.so.1'
checking sub-depends for 'not found'
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x00000000)
        libc.so => not found (0x00000000)
        not a dynamic executable
```

It might be a gcc version issue so I tried to use the old but compatible sdk found [here](http://archive.openwrt.org/barrier_breaker/14.07/ar71xx/generic/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2.tar.bz2) but it has no golang support.

I did try replacing feeds/packages with lastest sdk and run './scripts/feeds install -a' but with no success - golang tool chain is not compiled, and the building process failed at 

```
Finding targets
/home/mark/repo/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/feeds/packages/lang/golang//golang-build.sh: line 108: go: command not found

Building targets
/home/mark/repo/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/feeds/packages/lang/golang//golang-build.sh: line 129: go: command not found
```

I even tried to install go, but that did't work, maybe because it's cross platform compiling.

Although this is most likely not an openwrt-v2ray problem at all, would you please help with any suggestions?

Thank you!

kuoruan, 2021-02-22:
Try install with this command:

```
opkg --add-arch mipsel_74kc:20 install xxx.ipk
```

mkmark, 2021-02-23:
> Try install with this command:
> 
> ```
> opkg --add-arch mipsel_74kc:20 install xxx.ipk
> ```

Thank you, this indeed save me the trouble of adding the arch manually in opkg.conf, however couldn't solve my problem.

```
root@TP-LINK:~# opkg --add-arch mipsel_74kc:20 install v2ray-core_4.34.0-1_mips_24kc.ipk
Installing v2ray-core (4.34.0-1) to root...
Configuring v2ray-core.
//usr/lib/opkg/info/v2ray-core.postinst: line 5: default_postinst: not found
Collected errors:
 * pkg_run_script: package "v2ray-core" postinst script returned status 127.
 * opkg_configure: v2ray-core.postinst returned 127.
root@TP-LINK:~# v2ray
-ash: v2ray: not found
root@TP-LINK:~# ldd /usr/bin/v2ray
ldd: can't open cache '/etc/ld.so.cache'
checking sub-depends for '/lib/libgcc_s.so.1'
checking sub-depends for 'not found'
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x00000000)
        libc.so => not found (0x00000000)
        not a dynamic executable
```

As I reported, I first thought it was the arch or the cpu model problem too, but the ldd command told me that it is actually the c lib problem. Latest SDK all use the musl c lib, while the old OpenWRT 14.07 and its SDK uses uClibc lib.

That leaves me 2 choices, either I compile using the old SDK but somehow implant the Golang support, or I compile using the new SDK but somehow using the uClibc lib. I googled a lot but couldn't found reliable/easy solutions for both.

I am actually giving up since it's becoming too complex and am considering buying a new soft router...

Thank you for your wonderful work and all your support!

closed 2021-02-23 by mkmark

kuoruan, 2021-02-23:
If your router has enough space, you can install the offical file.

Try these files:

v2ray-linux-mips32.zip
v2ray-linux-mips32le.zip
v2ray-linux-mips64.zip
v2ray-linux-mips64le.zip