kuoruan/openwrt-v2ray #123: Cannot satisfy the following dependencies for v2ray-core-mini:

glenlau228, 2021-03-27:
Please attach the output of following commands.

```
cat /etc/*release
opkg print-architecture
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_TARGET='ar71xx/tiny'
DISTRIB_ARCH='mips_24kc'
DISTRIB_TAINTS='no-all'
DISTRIB_REVISION='R9.3.7 By Lean'
DISTRIB_DESCRIPTION='OpenWrt '
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://openwrt.org/"
BUG_URL="http://bugs.openwrt.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r1228-6dc35cf"
LEDE_BOARD="ar71xx/tiny"
LEDE_ARCH="mips_24kc"
LEDE_TAINTS="no-all"
LEDE_DEVICE_MANUFACTURER="OpenWrt"
LEDE_DEVICE_MANUFACTURER_URL="http://openwrt.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="OpenWrt SNAPSHOT r1228-6dc35cf"

Installing v2ray-core-mini (4.36.2-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for v2ray-core-mini:
 * 	ca-certificates * 
 * opkg_install_cmd: Cannot install package v2ray-core-mini.

glenlau228, 2021-03-27:
v2ray-core-mini_4.36.2-1_mips_24kc.ipk

kuoruan, 2021-03-27:
```
opkg update
opkg install ca-certificates
```

glenlau228, 2021-04-12:
安装上去了，但是遇到一个问题，启动不了，启动一下又变成了 服务未运行

kuoruan, 2021-04-12:
看一下日志报什么错误了。
另外，更新的时候可以先上传文件到 `/tmp/`, `tmp` 目录默认是使用的内存，这样就不用先卸载了。

glenlau228, 2021-04-12:
谢谢你的回复，我找了 配置的  /var/log/  目录中没有找到 v2ray 相关 的日志，

kuoruan, 2021-04-12:
可以看看系统日志

glenlau228, 2021-04-12:
Mon Apr 12 16:41:17 2021 daemon.info v2ray[6031]: V2Ray 4.37.2 (OpenWrt) R2 (go1.16 linux/mips)
Mon Apr 12 16:41:17 2021 daemon.info v2ray[6031]: A unified platform for anti-censorship.
Mon Apr 12 16:41:17 2021 daemon.info v2ray[6031]: 2021/04/12 08:41:17 [Info] main/jsonem: Reading config: /var/etc/v2ray/v2ray.main.json
Mon Apr 12 16:41:17 2021 daemon.info v2ray[6031]: main: failed to read config files: [/var/etc/v2ray/v2ray.main.json] > infra/conf: invalid field rule > infra/conf: failed to load geoip: private > infra/conf: failed to open file: geoip.dat > open /usr/bin/geoip.dat: no such file or directory
Mon Apr 12 16:41:17 2021 daemon.info procd: Instance v2ray::v2ray.main s in a crash loop 7 crashes, 18 seconds since last crash

提示未找到  failed to read config files: [/var/etc/v2ray/v2ray.main.json

kuoruan, 2021-04-12:
你装的是 mini 版本，不含 `geoip.dat`，但是你的配置文件里边用到了 `geoip.dat`，需要你手动下载一个 `geoip.dat` 文件，放到 `/usr/bin`

kuoruan, 2021-04-12:
到这里下：https://github.com/v2fly/geoip/releases

closed 2021-04-12 by kuoruan

glenlau228, 2021-04-12:
好的，感谢你的帮助。