kuoruan/openwrt-v2ray #128: openwrt 21.02.0-rc1，盼对应的release

yijing1998, 2021-05-16:
Please attach the output of following commands.

```
cat /etc/*release
opkg print-architecture
```
信息收集
```
root@OpenWrt-k2p-2:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="21.02.0-rc1"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.0-rc1"
VERSION_ID="21.02.0-rc1"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16046-59980f7aaf"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.0-rc1 r16046-59980f7aaf"

root@OpenWrt-k2p-2:~# opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10

root@OpenWrt-k2p-2:~# uname -a
Linux OpenWrt-k2p-2 5.4.111 #0 SMP Sun Apr 18 10:06:57 2021 mips GNU/Linux
```
报错信息
```
root@OpenWrt-k2p-2:~# opkg install /tmp/v2ray-core-mini_4.38.3-1_mipsel_24kc.ipk
Unknown package 'v2ray-core-mini'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core-mini found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package v2ray-core-mini.
```

其他
openwrt 19.07下，可以安装成功

kuoruan, 2021-05-16:
这是一个 OpenWrt 的 Bug，待修复
https://forum.openwrt.org/t/21-02-0-rc1-opkg-incompatible-with-the-architectures-configured/95222

kuoruan, 2021-05-16:
可以使用 `opkg install -V2 /tmp/v2ray-core-mini_4.38.3-1_mipsel_24kc.ipk` 来看看有没有错误的详细信息