kuoruan/openwrt-v2ray #130: Help on imagebuilder/repositories.conf

jakues, 2021-06-21:
I want install v2ray-core built-in on image openwrt
but shown `cant install v2ray-core`

I add this on repositories.conf
`src/gz kuoruan_packages http://openwrt.kuoruan.net/packages/releases/arm_arm1176jzf-s_vfp`

someone help me plz

jakues, 2021-06-21:
`Configuring ppp-mod-pppoe.
Collected errors:
 * opkg_install_cmd: Cannot install package v2ray-core.
make[2]: *** [Makefile:167: package_install] Error 255
make[1]: *** [Makefile:122: _call_image] Error 2
make: *** [Makefile:240: image] Error 2`

jakues, 2021-06-21:
`kuoruan_packages
Downloading http://openwrt.kuoruan.net/packages/releases/arm_arm1176jzf-s_vfp/Packages.sigSignature check failed.
Remove wrong Signature file.`

kuoruan, 2021-06-21:
You should add our key:

```
wget -O kuoruan-public.key http://openwrt.kuoruan.net/packages/public.key
opkg-key add kuoruan-public.key
```

jakues, 2021-06-21:
but Im using imagebuilder
since it cant use cmd : `opkg-key`

kuoruan, 2021-06-21:
OK, you should see our build guide:

https://github.com/kuoruan/openwrt-v2ray#custom-build

jakues, 2021-06-21:
solution only 

1. download .ipk $DIR
2. add `src v2ray-core file:$DIR`
3. then build 
thx for solutions

closed 2021-06-21 by jakues