kuoruan/openwrt-v2ray #152: to fix huge vssz OOM: Would you please to compile v2ray by Go 1.17.2 or later

wheelcomplex, 2021-11-02:
```
cat /etc/*release

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.0'
DISTRIB_REVISION='r16279-5cc0535800'
DISTRIB_TARGET='ath79/nand'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.0 r16279-5cc0535800'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="21.02.0"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.0"
VERSION_ID="21.02.0"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16279-5cc0535800"
OPENWRT_BOARD="ath79/nand"
OPENWRT_ARCH="mips_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.0 r16279-5cc0535800"

opkg print-architecture

arch all 1
arch noarch 1
arch mips_24kc 10
```

```
/usr/bin/v2ray -c /var/etc/v2ray/v2ray.main.json
or just
/usr/bin/v2ray --version
```
shows huge VSSZ memory usage, and also cause router to stop response.
```
3478 root      664m R    /usr/bin/v2ray --version
```
```
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 6450     1 root     S     665m 557%  29% /usr/bin/v2ray --config=/var/etc/v2ray/v2ray.main.json
```

some reports on the web:
https://github.com/shadowsocks/v2ray-plugin/issues/192

https://github.com/golang/go/issues/43699

And this should be fixed after Go 1.17.2:
https://github.com/golang/go/commit/05e6d28849293266028c0bc9e9b0f8d0da38a2e2

kuoruan, 2021-11-02:
https://github.com/kuoruan/openwrt-v2ray/runs/4033700518?check_suite_focus=true#step:7:2722

4.43.0-2 is already built with golang 1.17.2

kuoruan, 2021-11-02:
See the version of OpenWrt golang package:
https://github.com/openwrt/packages/tree/master/lang/golang

wheelcomplex, 2021-11-02:
Thanks, yes, it is 1.17.2, and I have compiled all of them from sources, but unfortunately it does not works, still ran into OOM and killed :(

wheelcomplex, 2021-11-23:
Just for some one who may need:  do not use geoip:cn tag in config will avoid v2ray to load geoip.dat which cause the huge memory usage during startup (and trigger OOM killer).

```
8554     1 root     S    42984  35%   4% /usr/bin/v2ray --config=/var/etc/v2ray/v2ray.main.json
```

PRoximaK, 2021-12-08:
I have the same issue as u. I tried setting variable to fix it, but it just worked at the beginning. Now it turned worse.