kuoruan/openwrt-v2ray #154: 如何在openwrt同时运行两个不同config的v2ray?

niclau, 2021-11-15:
目前我在openwrt 18.06.1上已经运行了一个v2ray用作透明代理，
现在想再运行一个v2ray用作中转服务器。
通过修改/etc/init.d/v2ray发现不好使，它只会运行config1的实例。

START=95
CONFIG1=/etc/v2ray/config1.conf #透明代理
CONFIG2=/etc/v2ray/config2.conf #中转
SERVICE_WRITE_PID=1
SERVICE_DAEMONIZE=1

start() {
  service_start /usr/bin/v2ray -config $CONFIG1
  service_start /usr/bin/v2ray -config $CONFIG2
}

stop() {
  service_stop /usr/bin/v2ray
}

手动在ssh下执行v2ray -config /etc/v2ray/config2.conf则能正常启动第2个实例，
请问该如何修改这个启动脚本，使其能启动两个实例?

wheelcomplex, 2021-11-15:
/etc/init.d/v2ray 干的事情很多，你不如直接在 /etc/rc.local 里启动第二个实例

niclau, 2021-11-16:
> 
> 
> /etc/init.d/v2ray 干的事情很多，你不如直接在 /etc/rc.local 里启动第二个实例

我有尝试在rc.local里启动第2个实例，但当openwrt启动时它没有启动，不知原因是什么。
/usr/bin/v2ray -config /etc/v2ray/config2.conf &

wheelcomplex, 2021-11-16:
不知道你的18.06.1是怎么回事，我在20版是可以的。
也许你可以试试 chmod a+x /etc/rc.local && cat /etc/rc.local 然后 sh /etc/rc.local  试试手动能否执行。

weiqi-chen, 2023-01-10:
给无法在rc.local添加开机启动项的人的参考。我刚刚从坑里爬上来了。
1. 首先给rc.local添加可执行权限。
```
chmod +x /etc/rc.local
```
2. 按下方修改rc.local。
把标准输出、标准出错重定向到`/dev/null`至关重要，否则v2ray是无法启动的。
```
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.

/usr/bin/logger 'rc.local executing.'

/usr/bin/v2ray -c /etc/config/v2ray.json > /dev/null 2>&1 &

exit 0
```