kuoruan/openwrt-v2ray #159: v2ray-core_4.44.0-1_aarch64_cortex-a53.ipk无法使用

fengbo731, 2021-12-10:
您好，下载了 v2ray-core_4.44.0-1_aarch64_cortex-a53.ipk

一下是机器的信息。
root@OpenWrt:~# cat /usr/lib/os-release
NAME="OpenWrt"
VERSION="19.07-SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07-SNAPSHOT"
VERSION_ID="19.07-snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="unknown"
OPENWRT_BOARD="ipq60xx/generic"
OPENWRT_ARCH="aarch64_cortex-a53_neon-vfpv4"
OPENWRT_TAINTS="no-all busybox override"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07-SNAPSHOT r11364-ef56c85848"

报错
root@OpenWrt:~# opkg install v2ray-core_4.44.0-1_aarch64_cortex-a53.ipk 
Unknown package 'v2ray-core'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package v2ray-core.

请问如何处理

Gastovski, 2021-12-19:
你的CPU构架是aarch64_cortex-a53_neon-vfpv4
你恐怕要“v2ray-core_xxxx_aarch64_cortex-a53_neon-vfpv4.ipk”这样一个安装包才行，但作者似乎没有提供

hunteryaol, 2022-06-06:
我也和你相同的CPU构架，研究了好长时间，发现在OPKG设置里加下面3句话就能装成功
arch all 100
arch aarch64_cortex-a53_neon-vfpv4 200
arch aarch64_cortex-a53 300