kuoruan/openwrt-v2ray #166: cannot use "quic-go doesn't build on Go 1.18 yet.

RuyeNet, 2022-04-01:
https://github.com/fw876/helloworld/issues/836

https://github.com/RuyeNet/OpenWRT-x86_64/runs/5781305742

/workdir/openwrt/dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.23.0/internal/qtls/go118.go:5:13: cannot use "quic-go doesn't build on Go 1.18 yet." (untyped string constant) as int value in variable declaration

ctrysbita, 2022-04-16:
But the openwrt upstream are using go 1.18 XD

zhizhouzh, 2022-04-21:
我是这么解决的，供参考：
```
             git clone https://github.com/openwrt/packages.git
             git log lang/golang/
             git reset --hard 9704e900 # go 1.17
             change dir to lede project
             rm -rf feeds/packages/lang/golang/
             cp -r ../github/packages/lang/golang/ feeds/packages/lang/ 
```

RuyeNet, 2022-04-23:
> 我是这么解决的，供参考：
>
> ```
>              git clone https://github.com/openwrt/packages.git
>              git log lang/golang/
>              git reset --hard 9704e900 # go 1.17
>              change dir to lede [...]

不管用，还是会报错，我用的是openwrt官方，不是lede

zhizhouzh, 2022-04-25:
> 

openwrt 应该更简单了。就是找到 golang 的仓库，然后 git reset 到 1.17 上再编译。
你说的还是报错还是原来的错误吗，如果是的话，那仔细检查一下。
我前面给的代码只是一个思路，不一定要完全 copy 这些命令。

RuyeNet, 2022-04-25:
现在问题好像已经被修复了，我用最新的源码编译，已经不报错了

RuyeNet, 2022-04-26:
cd $GITHUB_WORKSPACE/openwrt
mkdir temp
mv feeds/packages/lang/golang/ temp/
git clone https://github.com/openwrt/packages.git temp/packages
cd temp/packages/
git reset --hard 9704e900
cd $GITHUB_WORKSPACE/openwrt
cp -r temp/packages/lang/golang/ feeds/packages/lang/golang

closed 2022-04-26 by RuyeNet