kuoruan/openwrt-v2ray #167: 我应该装哪个版本啊

wifi315, 2022-04-04:
[image]

zhengzhou, 2022-04-05:
mipsel_24kc 这个架构的，可以用这个看一下 `opkg print-architecture`

wifi315, 2022-04-05:
> mipsel_24kc 这个架构的，可以用这个看一下 `opkg print-architecture`

这个也装不上

wifi315, 2022-04-05:
root@ImmortalWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10
root@ImmortalWrt:~#

wifi315, 2022-04-05:
root@ImmortalWrt:/tmp# opkg install v2ray-core_4.44.0-1_mipsel_24kc.ipk
Unknown package 'v2ray-core'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency ca-certificates for v2ray-c                                     ore
 * pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core found, bu                                     t incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package v2ray-core.

wifi315, 2022-04-05:
oot@ImmortalWrt:~# cat /etc/*release
DISTRIB_ID='Openwrt'
DISTRIB_RELEASE='18.06-SNAPSHOT'
DISTRIB_REVISION='22.03.17'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='Openwrt 18.06'
DISTRIB_TAINTS='no-all'
NAME="ImmortalWrt"
VERSION="18.06-SNAPSHOT"
ID="immortalwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="ImmortalWrt 18.06-SNAPSHOT"
VERSION_ID="18.06-snapshot"
HOME_URL="https://immortalwrt.org/"
BUG_URL="https://github.com/immortalwrt/immortalwrt/issues"
SUPPORT_URL="https://github.com/immortalwrt/immortalwrt/discussions"
BUILD_ID="r7967-cd1e95514e"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="ImmortalWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://immortalwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="ImmortalWrt 18.06-SNAPSHOT r7967-cd1e95514e"
root@ImmortalWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10
root@ImmortalWrt:~#

帮我搞一下吧

wifi315, 2022-04-05:
[新建文本文档.txt](https://github.com/kuoruan/openwrt-v2ray/files/8418463/default.txt)

zhengzhou, 2022-04-06:
看错误是缺少依赖，直接用readme里面安装源的方式可以连依赖一起安装，但是如果你这个搞不明白这个用不来的，这个是都是命令行的，建议你找其他的

ktechmidas, 2022-05-01:
Install ca-certificates (opkg install ca-certificates) then this issue will go away.

The error message is wrong about the architecture being the problem

wifi315, 2022-05-01:
> 安装 ca-certificates (opkg install ca-certificates) 然后这个问题就会消失。
> 
> 关于架构是问题的错误消息是错误的
> 
> （对不起，我只会说英语......但我今天遇到了这个问题）

怎么解决呢

GUNPLA, 2022-08-23:
这问题太简单，笑喷了
opkg update;
opkg install v2ray-core*.ipk;