kuoruan/openwrt-v2ray #171: 希望能增加mips_siflower架构的版本

Lidociane, 2022-06-06:
[image]

garysdevil, 2022-06-26:
[image]

Hope that we can have this version of mips_siflower arc

aspark, 2022-10-07:
https://github.com/aspark/openwrt-v2ray/releases/tag/v5.1.0-1
https://github.com/aspark/openwrt-v2ray/releases/download/v5.1.0-1/v2ray-core_5.1.0-1_mips_siflower.ipk
https://github.com/aspark/openwrt-v2ray/releases/download/v5.1.0-1/v2ray-core-mini_5.1.0-1_mips_siflower.ipk

Lidociane, 2022-10-08:
> https://github.com/aspark/openwrt-v2ray/releases/tag/v5.1.0-1
> https://github.com/aspark/openwrt-v2ray/releases/download/v5.1.0-1/v2ray-core_5.1.0-1_mips_siflower.ipk
> https://github.com/aspark/openwrt-v2ray/releases/download/v5.1.0-1/v2ray-core-mini_5.1.0-1_mips_siflower.ipk

谢谢这位大佬的编译