kuoruan/openwrt-v2ray #175: The source code of v5.1 cannot be build.

fasthorse, 2022-10-15:
make[2]: *** [Makefile:369: /home/martin/openwrt/openwrt-sdk-21.02.4-x86-64_gcc-8.4.0_musl.Linux-x86_64/b                uild_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.built] Error 1
make[2]: Leaving directory '/home/martin/openwrt/openwrt-sdk-21.02.4-x86-64_gcc-8.4.0_musl.Linux-x86_64/p                ackage/v2ray-core'
time: package/v2ray-core/full/compile#2.48#1.06#6.73
    ERROR: package/v2ray-core failed to build (build variant: full).
make[1]: *** [package/Makefile:116: package/v2ray-core/compile] Error 1
make[1]: Leaving directory '/home/martin/openwrt/openwrt-sdk-21.02.4-x86-64_gcc-8.4.0_musl.Linux-x86_64'
make: *** [/home/martin/openwrt/openwrt-sdk-21.02.4-x86-64_gcc-8.4.0_musl.Linux-x86_64/include/toplevel.m                k:223: package/v2ray-core/compile] Error 2

wukaisky, 2022-10-17:
Me too. openwrt 21.02, golang ver 1.17.12

touch /home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.configured_c7d1a8fae973ed799ab69ba7fc55b731
rm -f /home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.built
touch /home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/wukaisky/openwrt/21.02/tmp/go-build" GO_MOD_CACHE_DIR="/home/wukaisky/openwrt/21.02/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/v2fly/v2ray-core/v4" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/v2fly/v2ray-core/v4/main" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0" GO_BUILD_DIR="/home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.go_work/build" GO_BUILD_BIN_DIR="/home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/wukaisky/openwrt/21.02/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0=v2ray-core-5.1.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/wukaisky/openwrt/21.02/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/wukaisky/openwrt/21.02/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/wukaisky/openwrt/21.02/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0=v2ray-core-5.1.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/wukaisky/openwrt/21.02/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/wukaisky/openwrt/21.02/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.go_work/build" GOCACHE="/home/wukaisky/openwrt/21.02/tmp/go-build" GOMODCACHE="/home/wukaisky/openwrt/21.02/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/wukaisky/openwrt/21.02/feeds/packages/lang/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1665838969' -linkmode external -extldflags '-L/home/wukaisky/openwrt/21.02/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/wukaisky/openwrt/21.02/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-s -w -X github.com/v2fly/v2ray-core/v4.version=5.1.0 -X github.com/v2fly/v2ray-core/v4.build=R1 -X github.com/v2fly/v2ray-core/v4.codename=OpenWrt -buildid '1665838969' -linkmode external -extldflags '-L/home/wukaisky/openwrt/21.02/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/wukaisky/openwrt/21.02/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets
no required module provides package github.com/v2fly/v2ray-core/v4/main; to add it:
	go get github.com/v2fly/v2ray-core/v4/main

Building targets
no Go files in /home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.go_work/build

make[3]: *** [Makefile:369: /home/wukaisky/openwrt/21.02/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.built] Error 1
make[3]: Leaving directory '/home/wukaisky/openwrt/21.02/package/my_package/openwrt-v2ray'
time: package/my_package/openwrt-v2ray/full/compile#2.33#0.90#9.52
    ERROR: package/my_package/openwrt-v2ray failed to build (build variant: full).
make[2]: *** [package/Makefile:116: package/my_package/openwrt-v2ray/compile] Error 1
make[2]: Leaving directory '/home/wukaisky/openwrt/21.02'
make[1]: *** [package/Makefile:110: /home/wukaisky/openwrt/21.02/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wukaisky/openwrt/21.02'
make: *** [/home/wukaisky/openwrt/21.02/include/toplevel.mk:230：world] 错误 2

youland, 2022-10-18:
我这里也是这样的，编译最新的不成功。

```Makefile:365: recipe for target '/home/user/openwrt/openwrt/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.built' failed
make[2]: *** [/home/user/openwrt/openwrt/build_dir/target-x86_64_musl/v2ray-core-full/v2ray-core-5.1.0/.built] Error 1
make[2]: Leaving directory '/home/user/openwrt/openwrt/package/v2ray-core'
time: package/v2ray-core/full/compile#2.07#0.82#31.90
    ERROR: package/v2ray-core failed to build (build variant: full).
package/Makefile:114: recipe for target 'package/v2ray-core/compile' failed
make[1]: *** [package/v2ray-core/compile] Error 1
make[1]: Leaving directory '/home/user/openwrt/openwrt'
/home/user/openwrt/openwrt/include/toplevel.mk:228: recipe for target 'package/v2ray-core/compile' failed
make: *** [package/v2ray-core/compile] Error 2```

youland, 2022-10-18:
编译时，下载源码，改成下面这个就行了，我已编译成功。
git clone https://github.com/aspark/openwrt-v2ray package/v2ray-core

```rm -rf /home/user/openwrt/openwrt/tmp/stage-v2ray-core
touch /home/user/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.v2ray-core.full_installed
make[2]: Leaving directory '/home/user/openwrt/openwrt/package/v2ray-core'
time: package/v2ray-core/full/compile#66.07#31.64#50.10
make[1]: Leaving directory '/home/user/openwrt/openwrt'```

fasthorse, 2022-10-19:
> 编译时，下载源码，改成下面这个就行了，我已编译成功。 git clone https://github.com/aspark/openwrt-v2ray package/v2ray-core
> 
> ```
> touch /home/user/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.v2ray-core.full_installed
> make[2]: Leaving directory '/home/user/openwrt/openwrt/package/v2ray-core'
> time: package/v2ray-core/full/compile#66.07#31.64#50.10
> make[1]: Leaving directory '/home/user/openwrt/openwrt'```
> ```

The v2fly binary cannot run correctly, and quit when restarting v2fly.

youland, 2022-10-20:
> The v2fly binary cannot run correctly, and quit when restarting v2fly.

Yes,me too.
The v2ray is not running in openwrt,I have not found solution.
I download other v2ray.ipk,all can not running.