kuoruan/openwrt-v2ray #177: 编译报错提示go  module找不到

346673164, 2022-10-25:
Please attach the output of following commands.

```
cat /etc/*release
编译环境
CentOS Linux release 7.2 (Final)
Tencent tlinux release 2.2 (Final)
Tencent tlinux release 2.2 (Final)
```
使用lede的openwrt源码编译成功，clone本仓库至package后。build对应插件报错。
Finding targets
no required module provides package github.com/v2fly/v2ray-core/v4/main; to add it:
        go get github.com/v2fly/v2ray-core/v4/main

Building targets
no Go files in /data/workspace/lede/build_dir/target-mipsel_24kc_musl/v2ray-core-full/v2ray-core-5.1.0/.go_work/build

make[2]: *** [Makefile:369: /data/workspace/lede/build_dir/target-mipsel_24kc_musl/v2ray-core-full/v2ray-core-5.1.0/.built] Error 1
make[2]: Leaving directory '/data/workspace/lede/package/v2ray-core'
time: package/v2ray-core/full/compile#0.37#0.42#4.73
    ERROR: package/v2ray-core failed to build (build variant: full).
make[1]: *** [package/Makefile:116: package/v2ray-core/compile] Error 1
make[1]: Leaving directory '/data/workspace/lede'
make: *** [/data/workspace/lede/include/toplevel.mk:230: package/v2ray-core/compile] Error 2
是在是对makefile不熟，求教是什么原因

346673164, 2022-10-31:
看起来是makefile里面有些目录写的是v4   v4替换成v5后看好像ok了。
但是v2ctl好像编译失败，这快不是很了解。而且看应该是exclude了不知道为什么还要编译v2ctl。
目前注释了v2ctl相关， 可以编译通过。
另外，用的lede的源码编译的，可能会有一些差异

youland, 2022-11-02:
> 

编译通过后，能放在openwrt里使用吗？我的安装后用不了，跑不起来。

346673164, 2022-11-02:
> 

core可以 但是luci-app不行。 好像是和luci的版本关系。   打算放弃了。

youland, 2022-11-02:
> > 
> 
> core可以 但是luci-app不行。 好像是和luci的版本关系。 打算放弃了。

可否把v2ray.ipk发一份给我，我的luci好像可以用，我试一下能不能跑起来，谢谢！
邮箱：eW91bGFuZDgxNjhAZ21haWwuY29t
base64解码

346673164, 2022-11-17:
> 

我的已经删了。  对应ipk可以在官方的package里面下到的啊。